PandABlocks/PandABlocks-FPGA #14: Make FPGA project using Vivado

kimanha, 2019-09-04:
Hello,
 
I am trying to make compile, but I found some errors from my Linux environment.

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
/bin/sh: 1: ./common/python/parse_git_version.py: not found
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen
python2 -m common.python.generate_app /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/common/python/generate_app.py", line 9, in <module>
    require("jinja2")
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 943, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 829, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'jinja2' distribution was not found and is required by the application
Makefile:83: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen] Error 1

Do you have a full Vivado project for build PandABox firmware?
I create a Vivado project but also found some missing modules(IPs) because of some files generated by python?

# unknown modules from Vivado project and how to get HDL file?
ttlout_ctrl
lvdsout_ctrl
inenc_ctrl
outenc_ctrl
pcap_ctrl
system_cmd_fifo
soft_blocks

Araneidae, 2019-09-05:
```
/bin/sh: 1: ./common/python/parse_git_version.py: not found
```
I have an idea about this one.  Does your system have `/bin/env`?  I suspect not, and I think this is a bug in the script.  You can work around this problem by changing `/bin/env` to `/usr/bin/env` in the script.  I'll raise a separate issue for this, issue #15.

The message about `jinja2` not found is simple enough: this is a python package that is needed to build the firmware, so you'll need to install this on your build system.

kimanha, 2019-09-05:
>>You can work around this problem by changing /bin/env to /usr/bin/env in the script
Yes, I changed.

>>so you'll need to install this on your build system.
Installed.

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
**/usr/bin/env: ‘dls-python’: No such file or directory**

rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen
python2 -m common.python.generate_app /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini
####################################
# Resource usage
#  Block addresses: 24/32
#  Bit bus: 91/128
#  Pos bus: 19/32
#  Ext bus: 11/32
####################################
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/
echo building FPGA
building FPGA
make -C /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile VIVADO=/opt/Xilinx/Vivado/2015.1/settings64.sh \
            TOP=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA TARGET_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox BUILD_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ \
            IP_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
make[1]: Entering directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
rm -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'library ieee;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'use ieee.std_logic_1164.all;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'package panda_version is' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_VERSION: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_BUILD: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"81d5579b";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'end panda_version;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
source /opt/Xilinx/Vivado/2015.1/settings64.sh && vivado -mode batch -source /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/scripts/build_ips.tcl \
  -log build_ips.log -nojournal \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
/bin/sh: 1: source: not found
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile:46: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo' failed
make[1]: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo] Error 127
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
Makefile:207: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit] Error 2

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
/usr/bin/env: ‘dls-python’: No such file or directory
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen
python2 -m common.python.generate_app /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini
####################################
# Resource usage
#  Block addresses: 24/32
#  Bit bus: 91/128
#  Pos bus: 19/32
#  Ext bus: 11/32
####################################
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/
echo building FPGA
building FPGA
make -C /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile VIVADO=/opt/Xilinx/Vivado/2015.1/settings64.sh \
            TOP=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA TARGET_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox BUILD_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ \
            IP_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
make[1]: Entering directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
rm -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
[... cut ...]
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile:46: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo' failed
make[1]: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo] Error 127
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
Makefile:207: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit] Error 2

thomascobb, 2019-09-05:
Try pulling from master now, I have pushed some changes that should fix it.

If you haven't already, you need to set PYTHON in PandABlocks-FPGA/CONFIG to python rather than dls-python. You also need to make sure you have the numpy package installed as well as jinja2.

kimanha, 2019-09-05:
>>You also need to make sure you have the numpy package installed as well as jinja2.
Yes, already installed.

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen
python2 -m common.python.generate_app /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini
####################################
# Resource usage
#  Block addresses: 24/32
#  Bit bus: 91/128
#  Pos bus: 19/32
#  Ext bus: 11/32
####################################
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/
echo building FPGA
building FPGA
make -C /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile VIVADO=/opt/Xilinx/Vivado/2015.1/settings64.sh \
            TOP=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA TARGET_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox BUILD_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ \
            IP_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
make[1]: Entering directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
rm -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'library ieee;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'use ieee.std_logic_1164.all;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'package panda_version is' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_VERSION: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"32010100";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_BUILD: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"0fbc3f52";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'end panda_version;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
. /opt/Xilinx/Vivado/2015.1/settings64.sh && vivado -mode batch -source /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/scripts/build_ips.tcl \
  -log build_ips.log -nojournal \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
/bin/sh: 5: /opt/Xilinx/Vivado/2015.1/settings64.sh: source: not found
/bin/sh: 6: /opt/Xilinx/Vivado/2015.1/settings64.sh: source: not found
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile:46: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo' failed
make[1]: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo] Error 127
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
Makefile:208: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit] Error 2

I clearly see Vivado setting64.sh at path but some reason /opt/Xilinx/Vivado/2015.1/settings64.sh: source: not found.

here is my Vivado path.
bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ ls /opt/Xilinx/Vivado/2015.1/settings64.sh
/opt/Xilinx/Vivado/2015.1/settings64.sh

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ cat /opt/Xilinx/Vivado/2015.1/settings64.sh
##############################################################
# Copyright (c) 1986-2019 Xilinx, Inc.  All rights reserved. #
##############################################################

source /opt/Xilinx/SDK/2015.1/.settings64-Software_Development_Kit.sh
source /opt/Xilinx/Vivado/2015.1/.settings64-Vivado.sh

kimanha, 2019-09-05:
Can you please review my CONFIG ?
I changed only Vivado path from original CONFIG.example.

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ cat CONFIG
# Example configuration file for building PandA firmware, and for simulation
#
# Copy this file to a file named CONFIG and edit as appropriate.  Lines that are
# commented out can be left as they are, the default value is show.  Uncommented
# lines must be assigned values.

# Default build location.  Default is to build in build subdirectory.
# BUILD_DIR = $(TOP)/build

# Definitions needed for FPGA build
VIVADO = /opt/Xilinx/Vivado/2015.1/settings64.sh
#ISE = /dls_sw/FPGA/Xilinx/14.7/ISE_DS/settings64.sh
LM_LICENSE_FILE = 2100@APPS-LM4;

# Path to root filesystem
PANDA_ROOTFS = /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs
# MAKE_ZPKG = $(PANDA_ROOTFS)/make-zpkg

# Python interpreter for running scripts
#
# PYTHON = python2

# Sphinx build for documentation.
# SPHINX_BUILD = sphinx-build

# List of default targets to build when running make
# DEFAULT_TARGETS = zpkg

# FPGA Application Name
APP_NAME = PandABox-no-fmc

# vim: set filetype=make:

thomascobb, 2019-09-05:
Ok, looks like the shell for the makefile is not defaulting to bash.

Please can you pull from master and try again

kimanha, 2019-09-05:
bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
make: *** No rule to make target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini', needed by '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/autogen'.  Stop.

Copy PandABox-no-fmc.app.ini file from previous version.

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA_1/apps $ cp PandABox-no-fmc.app.ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/
echo building FPGA
building FPGA
make -C /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile VIVADO=/opt/Xilinx/Vivado/2015.1/settings64.sh \
            TOP=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA TARGET_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox BUILD_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ \
            IP_DIR=/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
make[1]: Entering directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
rm -f /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'library ieee;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'use ieee.std_logic_1164.all;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'package panda_version is' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_VERSION: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"36010100";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo -n 'constant FPGA_BUILD: std_logic_vector(31 downto 0)' \ >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo ' := X"0f46dfba";' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
echo 'end panda_version;' >> /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen/hdl/panda_version.vhd
. /opt/Xilinx/Vivado/2015.1/settings64.sh && vivado -mode batch -source /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/scripts/build_top.tcl \
  -log build_top.log -nojournal \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/ \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA//../autogen \
  -tclargs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo
WARNING: Default location for XILINX_VIVADO_HLS not found:

****** Vivado v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:07:21 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/scripts/build_top.tcl
# set TOP_DIR    [lindex $argv 0]
# set TARGET_DIR [lindex $argv 1]
# set BUILD_DIR  [lindex $argv 2]
# set AUTOGEN    [lindex $argv 3]
# set IP_DIR     [lindex $argv 4]
# set_param board.repoPaths $TARGET_DIR/configs
# create_project -force -in_memory panda_carrier_top \
#     $BUILD_DIR/panda_carier_top -part xc7z030sbg485-1
# set proj_dir [get_property directory [current_project]]
# set obj [get_projects panda_carrier_top]
# set_property "board_part" "em.avnet.com:picozed_7030:part0:1.0" $obj
# set_property "default_lib" "xil_defaultlib" $obj
# set_property "simulator_language" "Mixed" $obj
# set_property "target_language" "VHDL" $obj
# set_property part "xc7z030sbg485-1" [current_project]
INFO: [Project 1-154] The current project board 'em.avnet.com:picozed_7030:part0:1.0' is reset to empty (None). Use 'Project Settings' part selector (GUI) or issue set_property Tcl Command to set the board again.
# set_msg_config -id {[Synth 8-2644]} -suppress
# set_msg_config -severity "CRITICAL WARNING" -new_severity ERROR
# add_files $TARGET_DIR/const/panda-timing.xdc
# add_files $TARGET_DIR/const/panda-physical.xdc
# add_files $TARGET_DIR/const/panda-post_synth.xdc
# set_property used_in_synthesis false \
#     [get_files $TARGET_DIR/const/panda-physical.xdc]
# set_property used_in_synthesis false \
#     [get_files $TARGET_DIR/const/panda-post_synth.xdc]
# source $AUTOGEN/const/constraints.tcl
## read_ip $IP_DIR/fifo_1K32_ft/fifo_1K32_ft.xci
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2015.1/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'fifo_1K32_ft' generated file not found '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/ip_repo/fifo_1K32_ft/fifo_1K32_ft.dcp'. Please regenerate to continue.
[... cut ...]
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/Makefile:59: recipe for target 'panda_top.bit' failed
make[1]: *** [panda_top.bit] Error 1
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
Makefile:210: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/panda_top.bit] Error 2

glennchid, 2019-09-05:
Hi, can you try ' make ip_clean ' or ' make clean-all ' before running make? I suspect that you have a stale ip_repo directory from a previous failed build which is not being regenerated.

kimanha, 2019-09-05:
bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make clean-all
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/docs/build *.zpg
find -name '*.pyc' -delete

The zynq bitstram  "panda_top.bit" has been successfully generated.
Can you please see end of line that errors from slow FPGA build.

 FPGA Memory Resources User Guide for additional information.
INFO: [Vivado 12-3199] DRC finished with 0 Errors, 61 Warnings, 1 Advisories
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Writing bitstream ./panda_top.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-118] WebTalk data collection is enabled (User setting is ON. Install Setting is ON.).
INFO: [Common 17-186] '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Thu Sep  5 13:53:51 2019. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2015.1/doc/webtalk_introduction.html.
INFO: [Common 17-83] Releasing license: Implementation
write_bitstream: Time (s): cpu = 00:01:27 ; elapsed = 00:01:21 . Memory (MB): peak = 3009.699 ; gain = 165.922 ; free physical = 1673 ; free virtual = 25460
# write_hwdef -file $BUILD_DIR/panda_top_wrapper.hdf -force
# close_project
****** Webtalk v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:07:21 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/.Xil/Vivado-32476-bpmlab2-OptiPlex-7040/webtalk/labtool_webtalk.tcl -notrace
INFO: [Common 17-186] '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA/.Xil/Vivado-32476-bpmlab2-OptiPlex-7040/webtalk/usage_statistics_ext_labtool.xml' has been successfully sent to Xilinx on Thu Sep  5 13:53:55 2019. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2015.1/doc/webtalk_introduction.html.
INFO: [Common 17-206] Exiting Webtalk at Thu Sep  5 13:53:55 2019...
# exit
INFO: [Common 17-206] Exiting Vivado at Thu Sep  5 13:53:55 2019...
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/FPGA'
gcc -o tools/virtexHex2Bin tools/virtexHex2Bin.c
tools/virtexHex2Bin.c: In function ‘main’:
tools/virtexHex2Bin.c:27:29: warning: format ‘%x’ expects argument of type ‘unsigned int *’, but argument 3 has type ‘unsigned char *’ [-Wformat=]
     while(EOF!=fscanf(stdin,"%02x",&StreamData))

mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/SlowFPGA/

Creating bit map...
Saving bit stream in "slow_top.bit".
Bitstream generation is complete.
promgen -w -p hex -o slow_top.hex -u 0 slow_top.bit
Release 14.7 - Promgen P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
0x53394 (340884) bytes loaded up from 0x0
Using generated prom size of 512K
Writing file "slow_top.hex".
Writing file "slow_top.prm".
Writing file "slow_top.cfi".
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/tools/virtexHex2Bin < slow_top.hex > slow_top.bin
*** stack smashing detected ***: /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/tools/virtexHex2Bin terminated
Aborted (core dumped)
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/targets/PandABox/SlowFPGA/Makefile:68: recipe for target 'slow_top.mcs' failed
make[1]: *** [slow_top.mcs] Error 134
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/SlowFPGA'
Makefile:222: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/SlowFPGA/slow_top.bin' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/SlowFPGA/slow_top.bin] Error 2

kimanha, 2019-09-05:
I did make again and shows different errors.
../PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/SlowFPGA
I can see the slow_top.bit

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
python2 -m common.python.make_ipmi_ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/ipmi.ini
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/extensions
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/extensions
python2 -m common.python.make_extensions /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc.app.ini PandABox /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc/extensions
Adding extension system
sphinx-build -b html docs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/html
Running Sphinx v1.3.6
making output directory...
loading pickled environment... not yet created
loading intersphinx inventory from https://docs.python.org/2.7/objects.inv...
loading intersphinx inventory from http://PandABlocks-server.readthedocs.io/en/latest/objects.inv...
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 43 source files that are out of date
updating environment: 43 added, 0 changed, 0 removed
reading sources... [ 20%] build/fmc_acq427_doc
Exception occurred:
  File "/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/common/python/ini_util.py", line 18, in read_ini
    os.path.abspath(x) for x in sorted(errored)]
AssertionError: Can't read ini files [u'/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/modules/fmc_acq427/fmc_acq427_in.block.ini']
The full traceback has been saved in /tmp/sphinx-err-hXBhRn.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
Makefile:125: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/html' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/html] Error 1

glennchid, 2019-09-05:
The earlier error seems to be connected with the VirtexHex2Bin code for converting between bit and bin format, for the Slow FPGA (Spartan-6) configuration. The bit file generation completed successfully, but do you have a slow_top.bin file? Otherwise I do not know why the error was not reproduced the second time. I believe it may be possible to generate the binary file directly with bitgen, which would get around this problem - I will check tomorrow.

The second error appears to be connected with the documentation system - I will have to leave it to others to comment on this.

thomascobb, 2019-09-06:
I've fixed that particular docs error, could you try again please?

kimanha, 2019-09-06:
slow_top.bit and slow_top.bin has been generated.
But slow_top.mcs was not created.

glennchid, 2019-09-06:
I have pushed some changes which should fix the Slow FPGA configuration file issue. Please could you test this, either run 'make clean' or just delete the build/apps/.../SlowFPGA directory (if you don't want to rebuild the Zynq image), and rerun make. The file slow_top.bin should build without errors. There should no longer be an mcs file in the directory- this is only needed for directly programming a PROM (if present on the hardware), and can be easily generated using promgen and the bit or bin file.

kimanha, 2019-09-06:
>>The file slow_top.bin should build without errors. There should no longer be an mcs file ...
Yes, It work as you explained.

Please see end of page "make-zpkg" path error, I confirmed path is correct.
 
:
Using generated prom size of 512K
Writing file "slow_top.bin".
Writing file "slow_top.prm".
Writing file "slow_top.cfi".
make[1]: Leaving directory '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/SlowFPGA'
python2 -m common.python.make_ipmi_ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc-sfp1_eventr.app.ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/ipmi.ini
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
python2 -m common.python.make_extensions /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc-sfp1_eventr.app.ini PandABox /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
Adding extension system
sphinx-build -b html docs /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/html
Running Sphinx v1.3.6
making output directory...
loading pickled environment... not yet created
loading intersphinx inventory from https://docs.python.org/2.7/objects.inv...
loading intersphinx inventory from http://PandABlocks-server.readthedocs.io/en/latest/objects.inv...
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 43 source files that are out of date
updating environment: 43 added, 0 changed, 0 removed
reading sources... [100%] tutorials/tutorial4_snake_scan
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
WARNING: dot command 'dot' cannot be run (needed for graphviz output), check the graphviz_dot setting
writing output... [100%] tutorials/tutorial4_snake_scan
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/docs/build/pcap_doc.rst:17: WARNING: 'any' reference target not found: *PCAP.ARM=
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/docs/build/pcap_doc.rst:17: WARNING: 'any' reference target not found: *PCAP.DISARM
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/common/python/simulations.py:docstring of common.python.simulations.BlockSimulation.on_changes:None: WARNING: py:obj reference target not found: Dict[str, int]
generating indices... genindex py-modindex
highlighting module code... [100%] common.python.configs
writing additional pages... search
copying images... [100%] build/../../build/html/plot_directive/build/pulse_doc-8.png
copying static files... done
copying extra files... done
dumping search index in English (code: en) ... done
dumping object inventory... done
build succeeded, 4 warnings.
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs/make-zpkg -t /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA -b /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr -d /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build \
            etc/panda-fpga.list PandABox-no-fmc-sfp1_eventr-1.1-63-gc63ce01
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs/make-zpkg: 83: /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs/make-zpkg: 2: not found
Missing arguments: try -h for help
Makefile:263: recipe for target '/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/panda-fpga@PandABox-no-fmc-sfp1_eventr-1.1-63-gc63ce01.zpg' failed
make: *** [/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/panda-fpga@PandABox-no-fmc-sfp1_eventr-1.1-63-gc63ce01.zpg] Error 1

bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ ls 
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs/make-zpkg

thomascobb, 2019-09-06:
Same /bin/sh problem, but this time with make-zpkg.

I've pushed changes to the PandABlocks-rootfs repo, please update it and try again

kimanha, 2019-09-06:
bpmlab2 ~/xilinx/kiman/PandA/PandABlocks-FPGA $ make
python2 -m common.python.make_ipmi_ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc-sfp1_eventr.app.ini /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/ipmi.ini
rm -rf /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
mkdir -p /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
python2 -m common.python.make_extensions /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/apps/PandABox-no-fmc-sfp1_eventr.app.ini PandABox /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr/extensions
Adding extension system
/home/bpmlab2/xilinx/kiman/PandA/PandABlocks-rootfs/make-zpkg -t /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA -b /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build/apps/PandABox-no-fmc-sfp1_eventr -d /home/bpmlab2/xilinx/kiman/PandA/PandABlocks-FPGA/build \
            etc/panda-fpga.list PandABox-no-fmc-sfp1_eventr-1.1-63-gc63ce01

Thank you and appreciate for your support.
Next step, I am considering the rootfs when I getting the PanndaBox hardware.

closed 2019-12-05 by thomascobb

Liamgreat, 2020-08-09:
hello，I am trying to make compile, but I found some errors 。i use vivado2018.1 and i also change the scripts .but i find some errors.
[image]

thomascobb, 2020-08-10:
Unfortunately the build seems to be tightly coupled to the vivado version. We tried updating to vivado 2018.1 (the results are on the new-vivado branch), but never finished the job. You could have a look at the changes on that branch and see if any of those fix the problem.