PandABlocks/PandABlocks-FPGA #34: position_sign_extension is not optionnal

GTHIBAUX, 2020-10-13:
with Src version: PandA SW: 2.0.1 FPGA: 2.0.3C22 04bb2b28 04bb2b28 rootfs: PandA 2.0.1

We have tested with these relative encoders:
kubler,  bissc 12 bit ST + 12 MT > 24 bits (BINARY)
baumer,  BMMH 42S1G05C12/13B25 (seems to be BINARY with 1 sign bit)

The value returned is always signed and this could induce differences in value from the motor controler.
And majority of our relative encoders are BINARY (not signed).

Maybe we should add SIGNED enum in IN/OUTENC. ENCODING as done with encoder_gray branch?

thomascobb, 2020-10-15:
That sounds like a good idea, only sign extend if SIGNED=Yes

valerixb, 2020-10-19:
Hi
    maybe we should add another property like, for example, IN/OUTENC.SIGN that can be SIGNED or UNSIGNED? This allows to use different encodings (binary, gray) along with different sign choices (signed, unsigned) and all 4 the possibilities can be selected. Or maybe directly put 4 possibilities in the IN/OUTENC.ENCODING property.

thomascobb, 2020-10-19:
This raises an interesting question, how do we handle signed vs unsigned values at the moment? Most of the time they are passed straight through, but the pcap code treats them as signed:

https://github.com/PandABlocks/PandABlocks-FPGA/blob/4bb2b283ecf7b61503a3126a49714d5240d0d47c/targets/PandABox/hdl/pcap_frame_mode.vhd#L128

So this means that all positions are treated as 32-bit signed to PCAP. This means that 32-bit unsigned encoders won't be represented correctly in PCAP. We can't really do much about this as the averaging code is already pretty expensive.

I think I slightly prefer putting all the options into ENCODING, but if it's easier to have 2 fields then that is fine too.

thomascobb, 2021-04-01:
Closed via #39

closed 2021-04-01 by thomascobb