PandABlocks/PandABlocks-FPGA #41: Block SEQ: Errors in the sequence.

jbeca, 2021-02-22:


jbeca, 2021-02-22:
### oscilloscope : show errors in sequence.
[image]

### sequencer table :
[table.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/6023778/table.txt)

shu-soleil, 2021-03-18:
Jerome, could you try to set the SEQ1.TABLE with phase time > 1 (8ns) on each line ?

thomascobb, 2021-03-19:
I've reproduced this, I'll look into it...

shu-soleil, 2021-03-19:
L've made some short tests yesterday, it seems the SEQ could miss or jump lines when 'Phase 2 Time'=1.

thomascobb, 2021-03-22:
Yes, the problem is here:
https://github.com/PandABlocks/PandABlocks-FPGA/blob/f5798aff4f9b602c450f98e299a8d088e1f82bbb/modules/seq/hdl/seq.vhd#L276-L280

It looks like the last part of the `or` tries to pre-load in PHASE1 if PHASE2 is too short, but the conditions aren't restrictive enough. I'll try to find a more general fix.

As a workaround, if you make PHASE2 anything other than 1 clock tick then everything seems to work.

thomascobb, 2021-04-01:
Tested on a real PandA and it seems to work, so have merged into master

closed 2021-04-01 by thomascobb