PandABlocks/PandABlocks-FPGA #51: Add support for calculated registers

thomascobb, 2021-08-03:
As per https://github.com/PandABlocks/PandABlocks-server/issues/9

tomtrafford, 2021-08-12:
I was just having a quick look at how to implement this. I've made an example to how I understand it to work. For a block.ini such as:
```
[READ1]
type: read
description:
extension: test_extension1

[READ2]
type: read
description:
extension: test_extension1

[WRITE1]
type: write
description:
extension: test_extension2

[WRITE2]
type: write
description: 
extension: test_extension2

[READ3]
type: read
description:
extension: test_extension3

[READ4]
type: read
description:
extension: test_extension3

[WRITE3]
type: write
description:
extension: test_extension3

[WRITE4]
type: write
description: 
extension: test_extension3
```
A generated registers file would look like:

```
    READ1 READ2                    X test_extension1
    W WRITE1 WRITE2                X test_extension2
    READ3 READ4 W WRITE3 WRITE4    X test_extension3
```

So if multiple fields have the same extensions defined they will combine the entries in the Registers file depending on the field type. Have I understood this correctly?

Araneidae, 2021-08-12:
Hi @tomtrafford, this isn't right.  Have a look at https://github.com/PandABlocks/PandABlocks-server/blob/c8faaa58459743a975d363607d2979dbaf12108d/python/sim_config/registers#L246-L255 for an example register file.  The corresponding config file doesn't show the extensions at all.

The syntax of the extension register definition is described here: https://github.com/PandABlocks/PandABlocks-server/blob/c8faaa58459743a975d363607d2979dbaf12108d/server/extension.c#L238-L246:

Read registers:

    [register]* X extension-name

Write registers:

    [register]* [W [register]*] X extension-name

The meaning of this is: the first named registers will be read and passed to the extension server as arguments to the function `extension-name`.  If this is a write register then this function will be expected to return a tuple of values which will then be written to the registers after the optional `W`.

Hope this makes more sense.

Araneidae, 2021-08-12:
Hmm.  Why can't I link to lines of code anymore?

thomascobb, 2021-08-12:
@Araneidae I can't get my head around this.

Let's consider a simple example.

I want to have 2 write records, A and B, and 2 read records AVERAGE and DIFFERENCE. Whenever A or B is written to, the extension should calculate AVERAGE and DIFFERENCE:

```
    def write(self, block, a, b):
        average = int((a+b)/2)
        difference = a-b
        return (average, difference)
```

So if the FPGA register map was generated:
- A = 0
- B = 1
- AVERAGE = 2
- DIFFERENCE = 3

What would the registers file look like? We need to call the same write function in the extension each time A or B are changed, but set a different register number. My first guess was something like this:

```
BLOCK    29 transform
    A    0 1 W 2 3 transform
    B    0 1 W 2 3 transform
    AVERAGE 2
    DIFFERENCE 3
```

But how would the server know whether to write the value of A to register 0 or 1?

Araneidae, 2021-08-13:
I don't quite understand why you'd have a hardware implementation with 4 registers, so I'll work through a different example.

Let's have hardware where we want to specify an interval `[a,b]` by specifying the end points `a` and `b`, but we want the users to work with `centre` and `range`.  We'll use the extension server to do the translation.  So let's start with the `config` definition (for arguments sake, we'll have 10 instance of `RANGE` available to us):
```text
RANGE 10
    CENTRE    param
    RANGE     param
```
Note that the hardware registers don't appear.  Now to the `register` file:
```text
RANGE  29 interval
    CENTRE   W 0 1 X centre 
    RANGE    W 0 1 X range
```
Several things to notice here:

- Neither method reads anything from the hardware, so the state is going to be held within the server instance
- Two registers are written by both methods
- The registers don't have names
- Note that normally our PandA hardware registers are not read-write, they're either read-only or write-only.

So now let's implement this.  We'll need to create a module `interval.py` and ensure the extension server can find it.  The content of the file follows:
```python
# This class will be instantiated once when the `RANGE` register definition above is processed
class Extension:
    def __init__(self, count):
        self.ranges = [Range() for n in range(count)]
    def parse_write(self, field):
        if field not in ['centre', 'range']:
            raise ServerError('Invalid field name')
        def writer(block, value):
            return getattr(self.ranges[block], 'set_%s' % field)(value)
        return writer

class Range:
    def __init__(self):
        self.centre = 0
        self.range = 0
    def set_centre(self, centre):
        self.centre = centre
        return self._interval()
    def set_range(self, range):
        self.range = range
        return self._interval()
    def _interval(self):
        return (self.centre-self.range//2, self.centre+self.range//2)
```
While writing this it took me a while to remember that this is actually documented here: [Extension Server](https://pandablocks-server.readthedocs.io/en/latest/extension.html).  However, I think there is a bug in the documentation: the arguments to the `write_register` method are block number, value to write, followed by any hardware registers that may have been read.  In this example we're not reading any hardware registers.  This documentation error may be the cause of your confusion.

Also, now I come to write this, I realise that the extension server module interface is a bit clunky and could do with some rework.  In particular the postponing of the block number parameter to the actual read and write method means we can't just create an array of block instances and delegate everything to them.

Araneidae, 2021-08-13:
> there is a bug in the documentation

I have pushed a fix.

By the way, could we have a link to the online documentation on the front page?

Araneidae, 2021-08-13:
Perhaps a more reusable boilerplate version of the `Extension` class:
```python
class ExtensionHelper:
    def __init__(self, block_class, count):
        self.block_class = block_class
        self.blocks = [block_class(n) for n in range(count)]
    def parse_write(self, field):
        set_field = getattr(self.block_class, 'set_%s' % field, None)
        if set_field is None: raise ServerError('Cannot write to field "%s"' % field)
        def writer(block, value, *regs):
            return set_field(self.blocks[block], value, *regs)
        return writer
    def parse_read(self, field):
        get_field = getattr(self.block_class, 'get_%s' % field, None)
        if get_field is None: raise ServerError('Cannot read from field "%s"' % field)
        def reader(block, *regs):
            get_field(self.blocks[block], *regs)
        return reader
```
This could be injected into extension modules (in just the same way that `ServerError` is at the moment), and then an `Extension` module might look like this:
```python
class MyBlock: ...
def Extension(count):
    return ExtensionHelper(MyBlock, count)
```
Tempted to just add this to the extension server.

thomascobb, 2021-08-13:
ok, so to use Michael's example we need to make a block.ini to generate `config`:
```
INTERVAL 10
    CENTRE    param
    RANGE     param
```
and `registers`
```
INTERVAL  29 interval
    CENTRE   W 0 1 X centre 
    RANGE    W 0 1 X range
```
I suggest something like:
```
[TOP]
type: extension_write

[BOTTOM]
type: extension_write

[CENTRE]
type: param
extension: centre
extension_write: TOP BOTTOM

[RANGE]
type: param
extension: range
extension_write: TOP BOTTOM
```
So a new type `extension_write` that looks like a `write` record from the VHDL generation point of view, but doesn't have a `config` entry. Then a new `extension_write` entry that allows you to specify which fields with type `extension_write` should be written to. That should be enough for PandABlocks/PandABlocks-server#9 use case

What do you reckon @Araneidae ?

Araneidae, 2021-08-13:
Is the idea here that the `extension_write` type is a register that is only managed by writing from an extension?  I guess you'll also want an `extension_read` type.

thomascobb, 2021-08-13:
> Is the idea here that the `extension_write` type is a register that is only managed by writing from an extension?

Yes

> I guess you'll also want an `extension_read` type.

Probably, although I can't think of a sensible example at this point...

tomtrafford, 2021-09-02:
I've implemented this on a branch [calculated_registers](https://github.com/PandABlocks/PandABlocks-FPGA/tree/calculated_registers). This is only for the extension_writes at the moment, but I think I can add something similar for the extension_reads. I just wanted to check in your example above, when you have: `CENTRE   W 0 1 X centre` the 0 refers to TOP and the 1 refers to BOTTOM. Are 0 and 1 their Register numbers (what appears in addr_defines.vhd) or are they another number referring to each one?

Araneidae, 2021-09-02:
Hi @tomtrafford , I think in `CENTRE W 0 1 X`, we have to look at the associated python to see how the register assignments work (I've inlined the `self._interval` call here):
```python
    def set_centre(self, centre):
        self.centre = centre
        return (self.centre-self.range//2, self.centre+self.range//2)
```
The first returned value will be written to hardware register 0, the second to hardware register 1 (according to the `CENTRE W 0 1 X` definition), so I would say that `0` is `BOTTOM` and `1` is `TOP` here.

Oh: but I see that in @thomascobb's example he's written
```
[CENTRE]
type: param
extension: centre
extension_write: TOP BOTTOM
```
I think that's a (trivial) mistake, and the last line should be `extension_write: BOTTOM TOP` to match the example script.

Yes, 0 and 1 will be register numbers with corresponding names, which will only be visible to the VHDL; the server will never see the `BOTTOM` and `TOP` names.

thomascobb, 2021-09-06:
> I think that's a (trivial) mistake
That's right, I was thinking upside-down on the day...

closed 2023-09-20 by tomtrafford