PandABlocks/PandABlocks-FPGA #61: Add NumTrig and TrigCounter to PCAP

thomascobb, 2021-10-01:
Add 2 registers to PCAP:
- NumTrig: UInt32, if >0 then set ACTIVE=0 like falling edge of ENABLE
- TrigCounter: UInt32, Zero on rising edge of ENABLE, increment on each TRIG

thomascobb, 2021-10-01:
@tomtrafford do you have time to take a look at this?

tomtrafford, 2021-10-05:
Hi @thomascobb Yes I can have a look into this.

shu-soleil, 2021-10-05:
Hello, do you think it raisonable to take dma status into account for ACTIVE: ACTIVE=1 until dma finish the transferts then ACTIVE=0 ?

thomascobb, 2021-10-05:
I'm not sure, how much time difference is there typically between then it currently goes low, and when the DMA is complete?

shu-soleil, 2021-10-05:
> I'm not sure, how much time difference is there typically between then it currently goes low, and when the DMA is complete?

I made some tests, the different could be several ns or about 4 ms when the last 2 IRQ are very closed.

thomascobb, 2021-10-05:
And do you have a use case where keeping ACTIVE high an extra 4ms would be useful on the FPGA side? We always wait to do the next scan from when the server sends the completion code to the client, which means the DMA is complete. We use the ACTIVE to gate pulses on the input, so actually I'd prefer to keep the current behaviour of ACTIVE