PandABlocks/PandABlocks-FPGA #82: Move Kinja defs out of top_defines

coretl, 2022-07-01:
Make a new small top_defines_gen.vhd.jinja2 which contains just 

```
-- Bit Bus Width, Multiplexer Select Width -------------------------------
constant BBUSW                  : natural := 128;
constant BBUSBW                 : natural := 7;

-- Position Bus Width, Multiplexer Select Width.
constant PBUSW                  : natural := {{ total_pos_bus_length }};
constant PBUSBW                 : natural := 5;

-- Extended Position Bus Width.
constant EBUSW                  : natural := 12;
--------------------------------------------------------------------------

-- Presence of PCAP_STD_DEV functionality
constant PCAP_SUPPORTS_STD_DEV  : std_logic := '{{ ("pcap_std_dev" in fpga_options) | int}}';
```

Make another package and load it from top_defines

adhowell, 2023-02-03:
I'm not sure this is possible with the VHDL package system. E.g. if `top_defines` uses `top_defines_gen`, then the contents of `top_defines_gen` are only accessible in the scope of `top_defines`. VHDL does not have a proper linker system that would make the contents of `top_defines_gen` accessible to a file that only uses `top_defines`.

coretl, 2023-02-03:
@Araneidae I believe you suggested this, do you know if it is possible?

Araneidae, 2023-02-06:
If there are only a small number of symbols that need to be defined in `top_defines_gen` the simplest trick is to re-export them from `top_defines` thus:
```vhdl
use work.top_defines_gen.all;
package top_defines is
    -- These definitions have to be autogenerated
    alias PBUSW is work.top_defines_gen.PBUSW;
    alias PCAP_SUPPORTS_STD_DEV is work.top_defines_gen.PCAP_SUPPORTS_STD_DEV;

    -- All the rest go here
    ...
end;
```

closed 2023-02-23 by tomtrafford