PandABlocks/PandABlocks-FPGA #84: make single_hdl_test showed passed even when error occurs

shu-soleil, 2022-08-24:
Running 'make TEST="pcap 3" single_hdl_test' on the commit [2dc5f16](https://github.com/PandABlocks/PandABlocks-FPGA/commit/2dc5f16806772fe4a224a0feacde9fa38adbee66), the test result shows:
> The test result is pcap_3_tb
##################################### pcap_3_tb has passed #####################################
INFO: [Simtcl 6-16] Simulation closed

But actually a Fatal error occured
> FATAL_ERROR: Vivado Simulator kernel has discovered an exceptional condition from which it cannot recover. Process will terminate. For technical support on this issue, please open a WebCase with this project attached at http://www.xilinx.com/support.
Time: 156 ns  Iteration: 1

coretl, 2022-09-02:
@EmilioPeJu have you seen this before?

EmilioPeJu, 2022-09-04:
Have I seen Vivado crashing? yes, a few times, but unfortunate the logs don't show enough information to know the underlying condition... I stopped the simulation at time 155 ns and stepped line by line until crashing, it breaks somewhere in `pcap_buffer.vhd` inside process `ps_mode_bus` while trying to set `pcap_dat_o` using the record `mode_ts_bits`, my guess is that Vivado simulator is not handling records correctly under some unknown conditions... I've seen Michael having lots of problems with records a few times.

shu-soleil, 2022-09-05:
I've corrected the crash on the branch fixPCAPSimError, this issue is linked to PR #86 to add crash report on the testbench result.

closed 2022-10-07 by coretl