PandABlocks/PandABlocks-FPGA #86: Timing constraints not met for building PandABox-fmc_acq430 app on the Master

shu-soleil, 2022-09-02:
Timing constraints error reported while building PandABox-fmc_acq430 app on the Master (commit [2dc5f16](https://github.com/PandABlocks/PandABlocks-FPGA/commit/2dc5f16806772fe4a224a0feacde9fa38adbee66))
 
> '# set timingreport \
> '#     [report_timing_summary -no_header -no_detailed_paths -return_string \
> '#         -file post_route_timing_summary.rpt]
> '# if {! [string match -nocase {*timing constraints are met*} $timingreport]} {
> '#     send_msg_id showstopper-0 error "Timing constraints weren't met."
> '#     return -code error
> '# }
> ERROR: [showstopper-0] Timing constraints weren't met.
> ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.

shu-soleil, 2022-09-02:
Same error occured when building PandABox-fmc_acq427 app.

Troubles point to pll2_clkin1_out1, pll2_clkin2_out1 clocks routing.

shu-soleil, 2022-09-02:
[post_synth_timing_summary.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/9478290/post_synth_timing_summary.txt)
[post_place_timing_summary.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/9478306/post_place_timing_summary.txt)
[post_route_timing_summary.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/9478312/post_route_timing_summary.txt)

glennchid, 2022-09-03:
@EmilioPeJu - is there any reason not to register the signals 'ddr_a' and 'ddr_b' in finedelay.vhd?
These were the failing paths on at least one of the two failing builds and registering the signals solves the timing issues for these builds.

EmilioPeJu, 2022-09-04:
Sorry Glenn, I didn't notice the finedelay could introduce timing problems, feel free to register `ddr_a` and `ddr_b`, I don't think the extra latency in the output will be a problem at all.

Araneidae, 2022-09-05:
Oh, that's very interesting.  I wasn't expecting a timing problem here (all the signals are already on the correct clock domain, and the combinatorial logic is pretty simple, should be a single LUT per output), but I guess what's going on here is that the DDR inputs must have quite tight timing constraints.  Something to bear in mind for the future...

The obvious fix is to promote the combinatorial process to a clocked process (on clk_2x_i), but as there will be a phase shift in the result, the actual order of assignments in the statement will probably need to change.  Need to revisit the test bench and underlying logic.

I agree with @EmilioPeJu, an extra half tick (relative to the main clock) of latency here should be immaterial to the application.

EmilioPeJu, 2022-10-07:
@coretl , Done! it's in branch `fix-timing`, the testbench indicates it's working but at some point I'd like to test it physically with a pandabox and a scope.

closed 2022-11-03 by coretl