PandABlocks/PandABlocks-FPGA #87: Change SEQ.TABLE to be long table and add double buffering

coretl, 2022-09-30:
We currently have a 4k row limit on SEQ.TABLE, and get around this by having 2 SEQ blocks in a double buffered configuration. This wastes all our LUT blocks as we need to OR the outputs of SEQ1 and SEQ2 as well as putting logic around moving from one table to the next. With faster detector rates, we would like to push this double buffered approach into the SEQ itself. Potentially we will do this on the PGEN block in the future.

The plan:
- Change SEQ.TABLE to be long table, add SEQ.TABLE2 long table and SEQ.TABLE_NUMBER read uint
- Change logic to consume DMA blocks from TABLE until empty, then switch to TABLE2 if not-empty, reporting current TABLE_NUMBER
- Client side logic can fill the table that is not currently in use

This should be mostly backwards compatible although the behaviour of writing a table that is in use may be different. We shouldn't be doing that anyway.

@shu-soleil this will be in the 3.0 release, is this something that will affect soleil?

coretl, 2022-09-30:
Aim is to try this out in January

tomtrafford, 2023-02-22:
This is now to add another short table. Make both tables 2K (or 4K if possible?).

Araneidae, 2023-03-13:
@coretl, I would like to discuss this in person please.

closed 2023-06-15 by EmilioPeJu