PandABlocks/PandABlocks-FPGA #95: Add RateMeter mode for Counter module

shu-soleil, 2022-11-21:
To add RateMeter feature mode for the COUNTER module, I propose the changes on the featureCounterRateMeter branch which implicates:

- add a new param MODE and a new bit_mux RELOAD
- the default MODE will be "0: Normal" which performs continuous counting as usual
- "Normal" MODE counter will be reset to START if RELOAD rising edge
- new MODE "RateMeter": COUNTER.OUT is latched and updated when each RELAOD rising edge
- new MODE "Scaler": COUNTER.OUT is latched from ENABLE rising edge et updated when ENABLE falling edge

Would this be good for you ?

coretl, 2022-11-22:
I think the functionality is reasonable, but I'd like to see the requirements for the entire application to see if we can rationalize the bit_mux inputs. We have `TRIG` and `DIR`, and we can probably repurpose `DIR` to act as `RELOAD` if we think about it.

A sample usecase that I can think of:
- To have a DAC value output proportional to the count rate of a TTL input.

Could you give the use cases for your two new modes?

shu-soleil, 2022-11-23:
Do you use DIR bit of COUNTER? How about to keep RELOAD as but_mux and change DIR to enum param? 
Was DIR designed for pulse/dir counting ? In this case we could add PulseDir mode to QDEC module for this purpose.

COUNTER's new modes are to replace our CPCI counting card.
The most usecase is pulse counting for detectors as: point detector, intensity monitor (+VFC), photomultiplier (+discriminator) etc., in step-scan or free running monitoring.

coretl, 2022-11-25:
We have the same use case for point detectors and intensity monitors. The way we do it is:
- For captured data, set `COUNTER.OUT.CAPTURE=Diff`, then set the counter with `MIN=0` and `MAX=0`. Each trigger signal will give the difference between the counter at the beginning and end of the gate. So if you had a 1Hz trigger, you would get a counts per second reading each second.
- For free running, we use `FILTER.MODE=difference`, then trigger it at 1Hz. Then `FILTER.OUT` will give a counts per second.

Does this solve your use cases?

shu-soleil, 2022-11-29:
- For captured data we do it in the same way as you
- For free running, we tried also with COUNTER + FILTER, but in long time monitoring the Counter can be overflowed, if we reset COUNTER between each trigger then we'll need an extra SRGATE, that becomes too heavy to manage one counting channel. That's why we'd like to integrate this function in COUNTER module and potentially expand the DI channels with an FMC-DIO card.

coretl, 2022-11-29:
Does FILTER not cope with a rollover when MIN=MAX=0? If it doesn't then we should fix it. If this was fixed would it suit your free running case?

shu-soleil, 2022-11-29:
Do you mean to make FILTER take account of COUNTER.CARRY ? That requires adding a bit_mux to FILTER then... 
And our wish is to have at least 6 channels for free running counting, if we can manage 1 channel by one COUNTER module instead of 1 COUNTER + 1 FILTER that could much simply the config architecture.

coretl, 2022-11-29:
> Do you mean to make FILTER take account of COUNTER.CARRY ? 

If COUNTER rolls over from INT32MAX to INT32MIN then we shouldn't need the CARRY output, 32-bit arithmetic should just work. It does in PCAP without needing to take account of CARRY.

> And our wish is to have at least 6 channels for free running counting,

Ah, that makes sense. 6 FILTER blocks would be expensive.

Can you give a bit more information on the use cases of Scaler and RateMeter modes? I'd like to understand what the series of TRIG, RELOAD and ENABLE would be and what the OUT would be. Also, would you just be polling OUT to display on a GUI, or do you connect it to another block?

coretl, 2022-11-30:
How about this:
- Add a new param OUT_MODE
- In the default mode 0: "On change" then OUT will reflect the internal counter value each time it changes
- In the new mode 1: "On disable" then OUT will only be changed to the internal counter value on falling edge of ENABLE
- Use a PULSE block with NPULSES=0, WIDTH=(PERIOD - 1 tick), STEP=PERIOD wired to COUNTER.ENABLE

That would be the way I'd setup a RateMeter.
Would this work?

shu-soleil, 2022-12-02:
PULSE
I was thinking of this setup also but our TANGO device is very generic without much processing, a low level register is either static or passthrough to a TANGO attribut in most cases., it's not easy to set one PERIOD value to both PULSE and CLOCK modules (for PULSE triggering) and also to adjust PULSE.WIDTH dynamically with PERIOD's value. 

However how about to take your idea further, if we change PULSE to be able to generate pulses without external trigger but based on WIDTH + STEP instead?

coretl, 2022-12-02:
Sounds reasonable:
- Make PULSES=0 mean go forever
- Make TRIG=ENABLE=ONE start it off
- PERIOD changes as you suggest

shu-soleil, 2022-12-05:
After check it's good to only have infinite pulse number on free running mode. 
  
So as resume:

- add OUT_MODE param to COUNTER to reflect OUT 'on change' or 'on disable' (CARRY should be latched when 'on disable')
- add WIDTH param on CLOCK module, if PERIOD=0 then PERIOD <= (WIDTH+1 tick)

Do I forget something?

How do you prefer to manage if CLOCK's WIDTH >= PERIOD ? Rather to readjust PERIOD to (WIDTH+1 tick) as PULSE, or add a HEALTH read param to report error?

coretl, 2022-12-05:
>     * add OUT_MODE param to COUNTER to reflect OUT 'on change' or 'on disable' (CARRY should be latched when 'on disable')
> 
>     * add WIDTH param on CLOCK module, if PERIOD=0 then PERIOD <= (WIDTH+1 tick)

That sounds good.

> How do you prefer to manage if CLOCK's WIDTH >= PERIOD ? Rather to readjust PERIOD to (WIDTH+1 tick) as PULSE, or add a HEALTH read param to report error?

I think it's best to do the same for PULSE and CLOCK. I don't mind whether we readjust or add a HEALTH, as long as it's consistent.

coretl, 2022-12-05:
Maybe a better name for `OUT_MODE` would be `UPDATE_OUT`? And the enums could be "On trigger" and "On disable"?

shu-soleil, 2022-12-06:
> UPDATE_OUT

Seems better

> I think it's best to do the same for PULSE and CLOCK. I don't mind whether we readjust or add a HEALTH, as long as it's consistent.

Good, let's do same as PULSE

coretl, 2022-12-06:
Although maybe `OUT_MODE` is more extensible for the future...

I don't really know, I'll leave it to you to make a decision

shu-soleil, 2022-12-08:
Could you have a look at branches featureClockAddWidth and featureCounterRateMeter if they are good to you?

coretl, 2022-12-08:
I agree with the functionality but I'd like @glennchid or @Araneidae to review the VHDL. Please could you put 2 PRs in for those 2 branches?

closed 2023-02-03 by tomtrafford