PandABlocks/PandABlocks-FPGA #109: Need for a mechanism to check compatibility between FPGA and server

EmilioPeJu, 2023-02-27:
Michael pointed that we are introducing breaking changes in the next release and will need some way of checking that a certain server version will work with a certain FPGA bitstream version. He proposed that we create a new COMPAT register that is matched at the server.

Araneidae, 2023-02-27:
Specifically, my proposal is as follows:

- We add an extra register `DRV_COMPAT_VER` to the end of the `*DRV` block.  This is a change to both [PandABlocks-FPGA](https://github.com/PandABlocks/PandABlocks-FPGA) and [PandABlocks-server](https://github.com/PandABlocks/PandABlocks-server).
- We declare a driver protocol version number `DRIVER_COMPAT_VERSION` and set it to 1.  This value is then returned when reading the new `*DRV.DRV_COMPAT_VER` register.  The driver will refuse to load if this value is not read.

We then need to be a bit careful about managing these values.  An interesting idea is to add the definition of the driver compatibility version to the core `registers` file by adding a line of the form:
```ini
# Kernel driver compatibility version, used to ensure match with FPGA
DRIVER_COMPAT_VERSION = 1
```
This change would require changes in two places:

- The FPGA build process would need to either generate this line or load it from the existing `registers_server` file.

Araneidae, 2023-02-27:
Once we have the ability to define and read constants in the `registers` file we can also use this for blocks compatibility.  At the moment the contents of the `*REG` block are validated by the server, but any major protocol change cannot be detected.

I therefore suggest also adding the following line to the generated `registers` file:
```ini
# PandABlocks compatibility version
BLOCKS_COMPAT_VERSION = 3
```
This constant will be compiled into the server and cross-checked against any generated configuration at startup.

Araneidae, 2023-02-28:
This has been implemented in https://github.com/PandABlocks/PandABlocks-server/issues/31.

closed 2023-02-28 by Araneidae