PandABlocks/PandABlocks-FPGA #121: QDEC module removed

shu-soleil, 2023-04-14:
I tried to add an instance of QDEC in the no-fmc app, but the build fails with the error as below:
```
WARNING: [Vivado 12-818] No files matched '/home/informatique/eca/szhang/PandABlocks/V3/PandABlocks-FPGA/modules/qdec/hdl/*.vhd'
## set_property FILE_TYPE "VHDL 2008" [get_files $TOP_DIR/modules/qdec/hdl/*.vhd]
ERROR: [Common 17-55] 'set_property' expects at least one object.
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.

    while executing
"source $AUTOGEN/const/constraints.tcl"
    (file "/home/informatique/eca/szhang/PandABlocks/V3/PandABlocks-FPGA/common/scripts/build_top.tcl" line 61)
INFO: [Common 17-206] Exiting Vivado at Fri Apr 14 17:59:36 2023...
make[1]: *** [/home/informatique/eca/szhang/PandABlocks/V3/build/PandA-FPGA/BR_master/apps/PandABox-no-fmc-SOLEIL-3QDEC/FPGA/panda_top.bit] Error 1
make[1]: Leaving directory `/home/informatique/eca/szhang/PandABlocks/V3/build/PandA-FPGA/BR_master/apps/PandABox-no-fmc-SOLEIL-3QDEC/FPGA'
make: *** [fpga-bit] Error 2
```

glennchid, 2023-04-19:
This is because we are trying to set the "VHDL 2008" file type property to files which don't exist. The easiest solution is use the `-quiet` flag and have this fail quietly.

Araneidae, 2023-04-19:
I do have one obvious question here: why does the QDEC module have no VHDL files?

closed 2023-05-03 by tomtrafford