PandABlocks/PandABlocks-FPGA #125: Support the CAENels FMC-Pico-1M4

coretl, 2023-04-21:
@Araneidae is writing support for this, and it looks like it would be useful on beamlines to have support for this as it is substantially cheaper than a TetrAMM and we could synchronize it easier

Araneidae, 2023-04-24:
One implementation complication is worth noting.  The FMC readout must be clocked at a predictable frequency (over SPI) and so must be driven by a clock with a known frequency.  The current design of PandA now allows for multiple master clock frequencies, and the master clock at 125 MHz is a little slow if we want to drive the FMC at its highest capture frequencies.  Therefore we should probably use a fixed higher speed clock for the FMC interface.  I believe we already have a 200 MHz reference clock in the design, this would be suitable.

coretl, 2023-04-25:
The current ADC block takes a different approach, it is always capturing at 1MHz (I think it has a 200kHz low pass filter internally), then the value is captured and averaged by PCAP. The thought is that the clock rate is high enough with respect to the trigger rate (10kHz max) that we don't need to synchronize the clock rate, and can rely on synchronized signals to PCAP to give us Min/Max/Mean that corresponds to a single detector frame.

Do you think the same approach would work here?

For parameter naming, we use `GAIN1`..`GAIN8` enums:
https://github.com/PandABlocks/PandABlocks-FPGA/blob/f67231e2dead9e8a47f4de5bb14d00d6747bbb1f/modules/fmc_acq427/fmc_acq427.block.ini#L12-L20

And `VAL1`..`VAL8` for data:
https://github.com/PandABlocks/PandABlocks-FPGA/blob/f67231e2dead9e8a47f4de5bb14d00d6747bbb1f/modules/fmc_acq427/fmc_acq427.block.ini#L92-L96

The output is so 32-bit full scale gives +-10V `VAL`. Each step in `GAIN` shifts the output by one place, so the 5V gain will produce +-5V `VAL`

Araneidae, 2023-04-25:
Unfortunately we can't predict the actual gains for high and low gain for the FMC-Pico-1M4 as this will depend on the ordering options selected (high gain can range from 1 to 110 µA full scale, low gain from 0.5 to 20 mA full scale!)  Guess we'll have to name the `GAINx` options "Low" and "High".

Regarding free running operation I suggest tying `START=1` and just ignoring `STROBE` if not wanted.

For a first version I suggest we don't worry about reading calibration settings yet.