PandABlocks/PandABlocks-FPGA #129: Wrong initial value from PCAP capture

glennchid, 2023-05-16:
The first value read by PCAP corresponds to an earlier value from a previous acquisition, see below. Note the initial value (after power-up is some very large value and timestamp) but for subsequent acquisitions it will always give the N-6th value from the previous cycle (in this case at 12.9 ns):

```
arm_time: 2023-05-16T10:50:25.020Z
start_time: 2023-05-16T10:50:25.020Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.SAMPLES uint32 Value
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s

 3842715808 -1.520592637e+10
 69 0.300000024
 69 0.600000024
 70 0.900000024
 70 1.200000024
 70 1.500000024
 69 1.800000024
 69 2.100000024
 69 2.400000024
 69 2.700000024
 69 3.000000024
 69 3.300000024
 70 3.600000024
 70 3.900000024
 70 4.200000024
 70 4.500000024
 69 4.800000024
 70 5.100000024
 69 5.400000024
 69 5.700000024
 70 6.000000024
 70 6.300000024
 70 6.600000024
 70 6.900000024
 70 7.200000024
 69 7.500000024
 69 7.800000024
 69 8.100000024
 69 8.400000024
 70 8.700000024
 70 9.000000024
 70 9.300000024
 70 9.600000024
 69 9.900000024
 70 10.20000002
 70 10.50000002
 69 10.80000002
 70 11.10000002
 69 11.40000002
 69 11.70000002
 70 12.00000002
 69 12.30000002
 70 12.60000002
 69 12.90000002
 69 13.20000002
 70 13.50000002
 70 13.80000002
 69 14.10000002
 69 14.40000002
 69 14.70000002
END 50 Ok
arm_time: 2023-05-16T10:52:58.643Z
start_time: 2023-05-16T10:52:58.643Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.SAMPLES uint32 Value
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s

 69 12.90000002
 71 0.300000024
 71 0.600000024
 70 0.900000024
 70 1.200000024
 71 1.500000024
 70 1.800000024
 70 2.100000024
 70 2.400000024
 70 2.700000024
 70 3.000000024
 70 3.300000024
 70 3.600000024
 70 3.900000024
 70 4.200000024
 70 4.500000024
 70 4.800000024
 70 5.100000024
 70 5.400000024
 71 5.700000024
 70 6.000000024
 71 6.300000024
 70 6.600000024
 70 6.900000024
 71 7.200000024
 71 7.500000024
 71 7.800000024
 70 8.100000024
 70 8.400000024
 71 8.700000024
 70 9.000000024
 70 9.300000024
 70 9.600000024
 70 9.900000024
 70 10.20000002
 70 10.50000002
 70 10.80000002
 70 11.10000002
 71 11.40000002
 70 11.70000002
 70 12.00000002
 71 12.30000002
 70 12.60000002
 70 12.90000002
 70 13.20000002
 70 13.50000002
 71 13.80000002
 70 14.10000002
 70 14.40000002
 71 14.70000002
END 50 Ok

```
If needed I can help to set up a test on Panda-202 which demonstrates this.

adhowell, 2023-06-30:
I have a fix for this in the branch linked to this issue. Output of sequential captures now looks like:

```
arm_time: 2023-06-30T16:05:02.780Z
start_time: 2023-06-30T16:05:03.380Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.SAMPLES uint32 Value
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s

 62500000 7.2e-08
 62500000 1.000000072
 62500000 2.000000072
 62500000 3.000000072
 62500000 4.000000072
 62500000 5.000000072
 62500000 6.000000072
 62500000 7.000000072
END 8 Disarmed
arm_time: 2023-06-30T16:05:11.676Z
start_time: 2023-06-30T16:05:12.276Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.SAMPLES uint32 Value
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s

 62500000 7.2e-08
 62500000 1.000000072
 62500000 2.000000072
 62500000 3.000000072
 62500000 4.000000072
 62500000 5.000000072
 62500000 6.000000072
 62500000 7.000000072
END 8 Disarmed
```

The problem was that the first time the DMA state goes active during a collection an interrupt was being issued to the PS. I don't understand why, but I think this interrupt was causing some kind of de-sync issue where the write address the PL was using no longer matched the read address the PS was using to retrieve captured values.

coretl, 2023-07-04:
@EmilioPeJu you added these lines in https://github.com/PandABlocks/PandABlocks-FPGA/commit/4f08fccc586f6909a8f0ce6bac5e737b624ad256 back in March. This explains why we haven't seen these issues in production, as we haven't put 3.0 alpha anywhere. I think we need some PCAP DMA tests, do any of those exist?

EmilioPeJu, 2023-07-04:
@coretl , no dma tests exist, it's a problem that we had since the beginning and creating them would be a significant amount of work. 
@adhowell , thanks for identifying the problem, I believe there is a better way to deal with the problem (which is that the panda-server expects a switch in the buffer no matter the interrupt reason). If you remove those lines, the start time will be capture when there is a different reason (timeout or buffer full) and will be an imprecise time. Instead, I would change the driver in the part after capturing the timestamp ([panda_stream.c#L202](https://github.com/PandABlocks/PandABlocks-server/blob/942f28af74e8691dc184eb6cd5ea057b5c915746/driver/panda_stream.c#L202)), you could check if bit 0, 5 and 6 of the status are all zeros in which case the only reason for the interrupt was to capture the time and the function could just return (without switching buffers)

Araneidae, 2023-07-05:
If there is an issue with the server can we please raise an issue linking here.  I won't do this as I don't properly understand the conversation.  In particular it's not clear to me which part of this issue is in the FPGA and which part is in the driver.

EmilioPeJu, 2023-07-05:
@Araneidae , this issue can be fixed in different way, the problem is basically that the driver will advance to the next buffer when the interrupt is only about capturing the start time. When I implemented that feature in the FPGA, I didn't think it was appropriate to switch buffers unnecessarily but I forgot to not make the driver advance the block in that case. There are 2 obvious ways to fix it:
- In the server repo, changing the driver to not advance the block if it got only a start event
- In the fpga repo, making the start event to also switch the buffer.
My latest message was suggesting to fix it using the first approach.

Araneidae, 2023-07-05:
Oh, I see: you're saying that in the current situation the server picks up data from the old buffer.  Yes, that sounds like a fix for the driver.  I'll move this issue across then.

Closing in favour of https://github.com/PandABlocks/PandABlocks-server/issues/37.

closed 2023-07-05 by Araneidae