PandABlocks/PandABlocks-FPGA #167: Investigate mechanism to read built-in MAC addresses from Secure EEPROM on XU5 and XU6

glennchid, 2023-11-08:
Under petalinux u-boot reads MAC address from the Secure EEPROM and populates this in the device-tree <local_mac_address> field. In our build a random locally administered addresses is put in the device-tree by u-boot, which is later over-ridden in linux with the value in `/qspi/MAC` via `ifconfig`. The present work-around is to manually write the Enclustra-assigned MAC address to the QSPI.

I tried the following entry in the dts (copied from the petalinux generated dts):
```
&gem0 {
	local-mac-address = [ff ff ff ff ff ff];
};
```
but this did not work on its own, as more config is required in petallinux, see eg:
https://support.xilinx.com/s/question/0D52E00006hpQdiSAE/uboot-ethernet-mac-address-from-eeprom-petalinux-20202?language=en_US

EmilioPeJu, 2024-03-15:
There are 2 pull requests dealing with this: https://github.com/PandABlocks/PandABlocks-FPGA/pull/186 and https://github.com/PandABlocks/PandABlocks-rootfs/pull/49

EmilioPeJu, 2024-03-19:
Solved in the PRs I mentioned in my last comment

closed 2024-03-19 by EmilioPeJu