PandABlocks/PandABlocks-FPGA #168: ADD Block POSENC

AMMARAicha, 2023-11-28:
Hello,
I'm a new member of the ISAC group and I need a POSENC block for my project. After compiling, I received this output: panda-fpga PandABox-posenc_block-3.0a2-168-gf67231e-dirty. Previously, I successfully obtained a similar block for ADC. My question is: why isn't this block (module) available after installing PandA? Additionally, if I install a new firmware, will the last installed firmware be removed?
Thank You

coretl, 2023-11-28:
Please install version 3 of the rootfs and all the zpkgs and see if that fixes it.

AMMARAicha, 2023-11-28:
Okay i will do thank you   ,
 and when  I install a new firmware, will the last installed firmware be removed (acq427) How can I create both types of blocks!

coretl, 2023-11-28:
If you get the system to a working state with v3 firmware, then take your changes and apply them on top of PandABlocks-FPGA master branch, build a new zpkg, and then load just that zpkg

AMMARAicha, 2023-11-29:
Hello, 
so i tried to install firmware of acq427 with V3 i had same problem of firmware not installed and i put my fmc of acq427 in pandabox
[FIRMWARE NOT INSTALLED_acq427.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/13498770/FIRMWARE.NOT.INSTALLED_acq427.txt)

coretl, 2023-11-29:
can you do `ls -l /opt/share/panda-fpga/load-panda-firmware` please?

AMMARAicha, 2023-11-29:
root@panda:/opt/share/panda-fpga# ls -l /opt/share/panda-fpga/load-panda-firmware
-rwxr-xr-x    1 root     root          1554 Oct 24 14:09 /opt/share/panda-fpga/load-panda-firmware

AMMARAicha, 2023-11-29:
it works now i forgot to add slowfpga , so i added acq427 but i didn't understand how can i add a block of POSENC , thank you

coretl, 2023-11-29:
> i didn't understand how can i add a block of POSENC

To do this, check out the PandABlocks-FPGA repo master branch, copy the app.ini closest to what you want, insert 

```ini
[POSENC]
number: 1
```

then type make, and hope that it passes timing.

> why isn't this block (module) available after installing PandA?

PandABox FPGA is pretty much full, every extra block we add makes it harder to pass timing. POSENC is not commonly used so we agreed to remove it from the standard firmwares. You can however make a custom firmware to add it in as above, hope this works for you.

glennchid, 2023-12-01:
Does your app require the sfp_panda_sync module? If so, this now requires additional IP for synchonising the clocks on the PandAs. Assuming you have an account with Xilinx this can be downloaded from:
https://www.xilinx.com/cgi-bin/docs/ctdoc?cid=4d633f9b-6f82-4989-9dce-0875ea58ae89;d=xtp715-picxo-fracxo.zip
and then you just need to point the `EXT_IP_REPO` variable to the unzipped location.
If you are requiring panda_sync then you can simply remove the references to it in the app.ini file.

glennchid, 2023-12-01:
Hi Dmitri, I hadn't realised there was a new version available - v4.0 since 09/28/23. Seems that's the only version available now. We are using v3.2.2 which was the only version available before. From looking at the changelog it seems the main differences are support for Versal devices. Hopefully there is nothing that will affect the use on PandA. I will check the newer version at some point, when I next get back to looking at this, but maybe you could give it a try in the meantime and let us know if you see any problems?

dmgav, 2023-12-01:
Hi Glenn. So far I did not notice anything unusual with the Panda with custom-built FPGA. I'll let you know if I see any issues.

AMMARAicha, 2023-12-01:
hello, I came across these errors with the new version. 
8 Infos, 0 Warnings, 0 Critical Warnings and 2 Errors encountered.
place_design failed
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

    while executing
"place_design"
    (file "/home/informatique/eca/ammar/V3.0/PandABlocks-FPGA/common/scripts/build_top.tcl" line 93)
INFO: [Common 17-206] Exiting Vivado at Fri Dec  1 14:36:14 2023...
make[1]: *** [/home/informatique/eca/ammar/V3.0/PandABlocks-FPGA/common/fpga.make:157: /home/informatique/eca/ammar/V3.0/build/PandA-FPGA/BR_master/apps/PandABox-fmc_acq427_SOLEIL/FPGA/panda_top.bit] Error 1
make[1]: Leaving directory '/home/informatique/eca/ammar/V3.0/build/PandA-FPGA/BR_master/apps/PandABox-fmc_acq427_SOLEIL/FPGA'
make: *** [Makefile:269: fpga-bit] Error 2

I'm not sure if the issue is related to this specific version or another issue

glennchid, 2023-12-01:
Hi Ammaraicha,
Can you share more of the log please, as the original error message is not shown.

AMMARAicha, 2023-12-04:
Good Morning , yes of course 
[make_FPGA.log](https://github.com/PandABlocks/PandABlocks-FPGA/files/13543515/make_FPGA.log)

AMMARAicha, 2023-12-04:
this my .ini [PandABox-fmc_acq427-SOLEIL.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/13543763/PandABox-fmc_acq427-SOLEIL.txt)

Araneidae, 2023-12-04:
Let me drop that into the conversation so it's easier to discuss:
```
[.]
description:
    SOLEIL standard set of PandABox blocks with:
        - D-TACQ ACQ427 FMC module
        - PandA synchroniser on SFP3
target: PandABox
includes: common_soft_blocks.include.ini

[FMC]
module: fmc_acq427

[SFP3_SYNC]
module: sfp_panda_sync
site: sfp 3

[SEQ]
number: 2

[BITS]
number: 1

[CALC]
number: 2

[CLOCK]
number: 2

[COUNTER]
number: 8

[DIV]
number: 2

[FILTER]
number: 2

[LUT]
number: 8

[PCOMP]
number: 2

[PGEN]
number: 2

[PULSE]
number: 10

[SRGATE]
number: 8
```
One thing that might be worth doing is to open the synthesised design in Vivado and running "Report Utilization" on it.  We can then get a sense of where all the resources have gone!  This gives you a hierarchical table that you can explore.  You probably need to lose at least 2,000-5,000 LUTs from your design.

coretl, 2023-12-04:
That ini file both `includes: common_soft_blocks.include.ini` then declares more blocks of the same type (like PULSE, SRGATE, etc.). I don't even know what happens there, @tomtrafford do you know? Either way, I think it would be clearer if you dropped the `includes` line, then played around with the numbers of other blocks until you found something that fitted in the FGPA

tomtrafford, 2023-12-04:
If there is an entry for a block in the app.ini with the same name as a block in the includes.ini the entry in the app.ini overwrites the entry from the includes.ini (it doesn't add the number together). So in this case the includes has no effect, but removing it will help avoid any confusion.

AMMARAicha, 2023-12-14:
Hello, I'm trying to compile a new block but with version 2.1 using vivado 2015.2 I have this problem 
ERROR: [Vivado 12-172] File or Directory '/home/informatique/eca/ammar/V2/build_featureModuleQfilter/ip_repo/PICXO_FRACXO/PICXO_FRACXO.xci' does not exist.

I used the file https://www.xilinx.com/cgi-bin/docs/ctdoc?cid=4d633f9b-6f82-4989-9dce-0875ea58ae89;d=xtp715-picxo-fracxo.zip as you advised but I could not find this file PICXO_FRACXO.xci

glennchid, 2023-12-14:
PIXCO_FRACXO requires Vivado 2020-something onwards, and there should be no mention of it prior to v3.
Have you tried `make clean-all` as it might be due to a stale build file from an earlier v3 build?
Otherwise I would need to see more of the log to debug ...

AMMARAicha, 2023-12-22:
In Soleil, we use version 2.1, so I need to compile using this specific version. I managed to do it with version 3 before, but now I can't. Here's the log file. 
 
 
 
[make_log.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/13751284/make_log.txt)

Araneidae, 2023-12-31:
I also noticed two interesting errors in the first two lines of the log file:
```
/usr/bin/env: ‘python’: No such file or directory
/bin/bash: python: command not found
```
This almost certainly means that a couple of scripts in the build process will have failed to run: it looks like your system doesn't have `python` installed as an alias for `python3`, and there seem to be some scripts in the build process which don't respect the `PYTHON` setting in `CONFIG`.

Because there is a syntax error in `panda_version.vhd` and this file is autogenerated can you please post a copy of this file; this should provide some more clues.

@AMMARAicha, when you say "[at] Soleil, we use version 2.1", do you mean [PandABlocks-FPGA release 2.1](https://github.com/PandABlocks/PandABlocks-FPGA/releases/tag/2.1) dated April 2021?  If so, I think we're going to struggle to support this 2 1/2 year old version!

AMMARAicha, 2024-01-05:
yes [PandABlocks-FPGA release 2.1]  you find here :
[make_FPGA.log](https://github.com/PandABlocks/PandABlocks-FPGA/files/13840340/make_FPGA.log)
[panda_version.txt](https://github.com/PandABlocks/PandABlocks-FPGA/files/13840357/panda_version.txt)

Araneidae, 2024-01-05:
Here is `panda_version.txt`:
```
library ieee;
use ieee.std_logic_1164.all;
package panda_version is
constant FPGA_VERSION: std_logic_vector(31 downto 0)   := X"";
constant FPGA_BUILD: std_logic_vector(31 downto 0)   := X"af1e12a";
end panda_version;
```
Yeah.  That's definitely not going to build.

You'll have to dig into how this file was generated.

glennchid, 2024-01-08:
@AMMARAicha is this still an issue for you?
As @Araneidae mentioned, it looks like you don't have `python` on your system. You can either just create a link to `python3` or change the line 
`export VERSION := $(shell ./common/python/parse_git_version.py "$(GIT_VERSION)")`
to
`export VERSION := $(shell $(PYTHON) common/python/parse_git_version.py "$(GIT_VERSION)")`
in the top-level `Makefile`, which should solve your current issue but other scripts may have the same problem too.

Are you really using v2.1 at Soleil? Any reason not to upgrade to v3? There have a lot of major changes to the codebase in the meantime, including bug fixes, and as @Araneidae mentions, it would be very difficult to support older versions.

AMMARAicha, 2024-01-09:
We're using v2.1 because it's the operational version at soleil. Yes, I managed to compile the app, but I used Python 2 as mentioned in the config.example. Additionally, I successfully compiled it with V3 PandABlocks. Thank you for your help

closed 2024-03-14 by tomtrafford