PandABlocks/PandABlocks-FPGA #170: Missing 'vio_1' IP while building `PandABox-no-fmc-sfp3_dls_eventr`

dmgav, 2024-01-11:
We are attempting to build `PandABox-no-fmc-sfp3_dls_eventr` app, and the built fails because of missing `vio_1` IP:

https://github.com/PandABlocks/PandABlocks-FPGA/blob/d28645efd13de606af7c6dda2f47e09b2efdb018/modules/sfp_eventr/sfp_dls_eventr.block.ini#L8

Is it an external IP? Could you please point us to the source where we can download it?

glennchid, 2024-01-15:
@dmgav, I have removed the debug core which was preventing the EVR from building, and pushed this to the master branch. Please can you try again now? Thanks, Glenn.

dmgav, 2024-01-16:
Thank you, I was able to successfully build zpkg. Do I understand correctly that the Panda EVR will still not work if connected directly to EVG? Do we need to add to the system to make it work? Should we use a standard MRF EVR card to recover the clock signal?

Eventually we would like to be able use MRF timing to get accurate absolute timestamps for collected data points. I understand that this would involve using recovered clock to generate relative timestamps and recording an absolute MRF timestamp for the event of PCAP activation. I am not fully familiar with Panda architecture. How difficult would be to implement this? NSLS-II has very similar timing system and the EVR implementation may potentially be shared. We are planning to work on those features and potentially contribute to the project.

closed 2024-03-14 by tomtrafford