PandABlocks/PandABlocks-FPGA #178: Autogenerate `system.block.ini` for different targets/platforms

glennchid, 2024-02-20:
At the moment we have a different System block for the two different hardware platforms, eg zynq(zynq-7000) and zynqmp (zynq ultrascale-plus). For selecting the hardware timestamp from the SFPs (see #172), the system block will need to be aware of the number of SFPs supported by the carrier. Rather than having a separate `system.block.ini` for each target, can we think of a way to autogenerate this file based on the information in the `target.ini` file? Is the use of jinja tempating of .ini files against the philosophy of the framework? Is there an easier or neater way to do this?

coretl, 2024-02-26:
I don't think the small amount of duplication here is worth the framework effort. There are other things in the system block like temperature sensors, so I think it makes sense for it to be target specific anyway

glennchid, 2024-02-27:
I was thinking more about the case where we have MGTs going to an FMC, such as for PandASwitch. In that considered an app for the UltraPandA or a separate target altogether?

coretl, 2024-02-27:
We could say that only the onboard 4x SFPs can clock the fabric or provide timestamps for UltraPandA, then no changes to SYSTEM are needed if we fit the SFP FMC

closed 2024-03-14 by tomtrafford