PandABlocks/PandABlocks-FPGA #179: PCAP.TS_START not zeroed when GATE already high at the start of capture

coretl, 2024-02-29:
If `PCAP.GATE` and `PCAP.ENABLE` are both tied to `ONE`, like in this diagram:
[image]

then `TS_START` on the last frame will be leftover from the previous scan rather than being 0:

```
$ nc 172.23.252.201 8889

OK
arm_time: 2024-02-29T17:31:03.969Z
start_time: 2024-02-29T17:31:03.970Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s
 PCAP.TS_END double Value scale: 8e-09 offset: 0 units: s

 0 0.500000064
 0.500000064 1.500000064
 1.500000064 2.500000064
 2.500000064 3.500000064
 3.500000064 4.500000064
END 5 Disarmed
arm_time: 2024-02-29T17:31:10.377Z
start_time: 2024-02-29T17:31:10.377Z
missed: 0
process: Scaled
format: ASCII
fields:
 PCAP.TS_START double Value scale: 8e-09 offset: 0 units: s
 PCAP.TS_END double Value scale: 8e-09 offset: 0 units: s

 4.889428296 0.500000064
 0.500000064 1.500000064
 1.500000064 2.500000064
 2.500000064 3.500000064
 3.500000064 4.500000064
END 5 Disarmed
```

tomtrafford, 2024-05-13:
I've added a timing test in [pcap-TS_START-test ](https://github.com/PandABlocks/PandABlocks-FPGA/tree/pcap-TS_START-test). However, I cannot replicate this behaviour in the VHDL. @EmilioPeJu Does a change need to be made in the server similar to https://github.com/PandABlocks/PandABlocks-server/issues/37 ?

EmilioPeJu, 2024-05-13:
The relevant change to the FPGA already happened when the start time feature was added. Was this bug introduced after that? if you don't see the problem in the timing tests, it might be a hardware problem that needs to be investigated with chipscope (or similar)

EmilioPeJu, 2024-06-28:
It would be nice simplifying that code, though that would require more work(and testing) than just fixing this bug

tomtrafford, 2024-07-01:
I've tried putting these changes onto a PandA but it does not seem to have fixed this problem.

EmilioPeJu, 2024-07-01:
Thanks for testing it, I have added one extra line that will (hopefully) fix it

tomtrafford, 2024-07-01:
I've just put the new changes onto a PandA. It still hasn't fixed this.

EmilioPeJu, 2024-07-01:
is it showing the same behaviour? I'm surprised, as soon as enable goes low, it should be clearing ts_start