PandABlocks/PandABlocks-FPGA #180: Investigate using SI5338 for MGT clocking on PandABrick

glennchid, 2024-03-01:
Compare iBERT tests for the PandABrick SFP using both the Si5338 clock generator on the ST1 and the AX3 oscillator on the FMC board. Also, compare PandASync app in both cases.

Will need to programme the oscillator via i2c from the PS.

gi4you, 2024-03-07:
xu5_st1 project used MERCURY XU5 Design-In Kit? correct model number and web link?

coretl, 2024-03-08:
Parts:
- ME-XU5-4EV-1I-D11E-R1.2 
- ME-ST1-W-R1 

Ordered separately, but the XU5 design in kit seems the same.

We are going to drop support for this platform though, as the XU5 doesn't wire its SFPs to the onboard connector on the ST1, it wires to the FMC. We need a custom FMC with an SFP on it in the PandABrick to make this work

Why do you ask?

gi4you, 2024-03-08:
Ok, thank you.
I saw Panda already support "xu5_st1-fmc_acq430.app" for a low cost zynq-Ultrascal+ high resolution ADC project.
And it is an interesting project for evaluation with the Ultrascal+ FPGA.
Then, are you considering other Ultrascal+ FPGA platforms?