PandABlocks/PandABlocks-FPGA #200: changing/giving MAC-addresse

nilsbuc, 2024-08-15:
I'm currently working on two PandAboxes. While setting the PandAbox up I couldn't get a connection via the serial-port at the back to enter a MAC-address, but they give you also the option to write the MAC-address inside a file (MAC.txt) and put this file onto SD-card from which the PandAbox boots, which I did.
I wrote the MAC address like this: XX:XX:XX:XX:XX:XX

The first PandAbox won't take the MAC-address and therefore won't boot. 
What did I do wrong?

The second PandAbox took the MAC-address (I don't know what I did different) but at that time I took a MAC-address from another pc in my network. So now that I want to introduce the PandAbox into the network, it obviously won't work. But putting another MAC-address inside the MAC.txt file won't change the MAC-address of the PandAbox. 
How can I change the MAC address?

tomtrafford, 2024-08-15:
I would recommend setting up the serial connection to monitor what is going on. There will be error messages explaining what is currently happening on the PandABox. There is a chance there is another problem which is causing it not to boot. For your first PandA which isn't booting, what are the LEDs on the front panel doing when booting? 

On Rootfs install the PandA will set its MAC address from a file called `MAC` on the SD Card. So I believe you will need to delete the files in the boot partition of the SD card, replace them with the contents of the boot.zip file and have the MAC-address in a file called `MAC` to change the MAC address on your second PandABox.

nilsbuc, 2024-08-16:
For the second PandAbox, I now took the SD-card, which was already repartitioned by the PandAbox, then deleted the files in boot partition and then put the content of the boot.zip file in the boot partition with a MAC.txt file (as seen in the screenshot). But the PandAbox still takes the old MAC-address.
[image]

coretl, 2024-08-16:
The MAC address storing procedure consists of:
- Checking `/qspi/MAC`
- If it exists then use that
- If it doesn't, then create the MAC from a file on the SD card called `MAC`, or prompt the user for it, storing it in `/qspi/MAC`

https://github.com/PandABlocks/PandABlocks-rootfs/blob/ef47480ae01aa5753a9bc446e43caf77b7e6d9d2/initramfs/init#L120-L125

> but they give you also the option to write the MAC-address inside a file (MAC.txt) and put this file onto SD-card from which the PandAbox boots, which I did.

I can't see any reference to `MAC.txt` in the code, only `MAC`. Where did you see this information?

> I wrote the MAC address like this: XX:XX:XX:XX:XX:XX

If creating a file `MAC` on the SD card, PandA is somewhat picky about the format, it must be an 18 character text file, with format `XX:XX:XX:XX:XX:XX` and a single trailing newline. 

> I thought, that when the "ERR"-LED flashes, the PandAbox is ready for the .zpkg to be installed (this was the case with the second PandAbox. But the SD-card isn't even repartitioned, so I guess the PandAbox is waiting for a MAC-address.

Unfortunately both states (no MAC address, and no .zpkg) are communicated to the user with the same LED sequence, blinking ERR and off STA. The only way to tell the difference is with the serial console.

> The serial connection won't give me any data, literally nothing (I'm using minicom to monitor what's happening).

Are you using a crossover (null modem type) serial cable?

> But putting another MAC-address inside the MAC.txt file won't change the MAC-address of the PandAbox.
> How can I change the MAC address?

Unfortunately the only way to set a different MAC address is to remove or edit the `/qspi/MAC` file in place. To do this you either need the serial console connection, or to SSH in.

In your case if you can't get the serial console working, I suggest you unplug the second PC from the network, plug the PandA in, let it DHCP and get the same address, use the web console to put your SSH keys on it, then `ssh root@<panda-ip> rm /qspi/MAC`, turn it off, blank the SD card, and put a new set of rootfs files, including a `MAC` file. When you put the SD card back in and boot it should reinstall itself including using the new MAC address.

nilsbuc, 2024-08-20:
OK, the problem really just was, that I had a MAC.txt file and the PandAbox wanted a MAC file. Now both PandAbox work.
Thank you very much.

closed 2024-08-20 by nilsbuc