PandABlocks/PandABlocks-FPGA #203: How to get the input with the FMC_ACQ430

nilsbuc, 2024-08-21:
Hey,
I have a signal generator hooked up to the fmc_acq430 module with a 5Vpp 50kHz sine-wave. 
I have no idea how I can read the value inputted into the fmc_acq430 and there aren't many blocks to which I can connect the val outputs. 
Are there more block librarys or do I have to create my own blocks? I'm pretty sure I have to this anyway some time in the future but I would like to skip it for now, cause I only want to see, that I can capture some input.
Also if there are more blocks, how do I get the .ini files onto the panda?

coretl, 2024-08-21:
Is the value scaled correctly and appearing in `http://<panda-ip>:8008/gui/PANDA/positions`? Here is a screenshot from a PandA that doesn't have that card, but it should appear in the list somewhere:

[image]

nilsbuc, 2024-08-27:
Yes, I think I have I have the values scaled correctly, but there is still nothing on the val output. I also put the Signal into an oscilloscope to be sure that the signal is correct, but everything seems fine.
I even tried other signals, like square and pulse, but there's also nothing.
[image]
[image]

nilsbuc, 2024-08-27:
I tried to add the value of two different sin signals (one with 5 Vpp and one with 10 Vpp). I captured val5 and val8, cause that are the two inputs I use and also CALC1.OUT for the result. But as you can see, I don't even got any Input from the FMC_ACQ430_IN.
I also tested the ttl of the FMC_ACQ430_IN, which works, so I don't think, that it's broken.
[image]
[image]

nilsbuc, 2024-08-28:
It works now, but I just took the Module out and put it back in and then restartet the PandA. Now I just have a lot of Noise on the inputs. 
[image]

closed 2024-08-30 by nilsbuc