PandABlocks/PandABlocks-FPGA #227: Prototype replacement of python sims with VHDL

coretl, 2025-04-03:
The python simulations aim to provide a realtime simulation of mostly idle blocks. E.g. a pulse block that makes 10Hz pulse outputs spends only ~10 ticks of every 125000000 actually doing something, so the python simulation is structured so that it jumps between the timestamps where it actually does things. However, maintaining these python simulations in parallel is painful. We would like to replace the python simulations with VHDL equivalents. This would require changing the architecture of the runtime VHDL block so they support both the runtime "every timestamp" processing, and the simulation "jumping timestamps" processing.

A way we could do this is to remove the `clk` input to the VHDL block that ticks every 8ns, then having a `process on rising edge of clk` statement that increments a counter and does an action when the counter reaches a particular value, we would have a `ts` input that incremented every clock tick, then have a `process on change of ts` statement that does an action when ts reaches a particular value.

At runtime (in the FPGA) we would increment `ts` and process the block every clock tick.

In simulation (under cocotb) we would only process the block if one of its inputs changed, or it had requested to be called at a particular `ts`, jumping the `ts` and only running the simuiation for one tick.

The support this we would need to change each block to work with `ts` instead of `clk`, and also provide a `next_ts` output that would only be used in simulation to declare when it needs a callback.

This ticket is to implement a single prototype of this by rewriting the pulse block in this manner and demonstrating the cocotb code that would do this.

Araneidae, 2025-04-04:
I would like to see an example of what you have in mind, but I don't think that what you are proposing is synthesisable.  Essentially it sounds like you're trying to make `ts` behave like a clock derived from `clk`, and you can't do this.  It is a classic mistake to try and treat a clock enable signal (your `ts` has changed event) like a clock.

I don't think you can do what you're trying to do here.

glennchid, 2025-04-04:
I agree that this sounds like some form of clock gating, which in general is not a good idea. It *might* be possible though, if done properly, for example with a BUFGCE. I guess the idea is that the gating logic would get optimised away by the synthesiser, leaving the clocking network looking like it does currently.

I am not sure I understand the intention well enough though. Would it not be better to do this clock gating in the simulation test-bench, or is this a workaround for some deficiencies of cocotb?

Araneidae, 2025-04-04:
I have chatted to @coretl about this, and as I understand it he would like to engineer VHDL blocks that only generate output transitions in direct response to inputs, in other words, there are no "hidden" state transitions.  This would then allow the simulation to skip "unnecessary" clock ticks.

This is, I think, difficult and extremely invasive to achieve, but might be doable.

We discussed counters, which are the most obvious form of "hidden" internal state.  These could be externalised by creating a simple counter entity which would be instantiated externally to the target block; this would allow the simulation to observe the start and completion of counters and therefore bypass the actual counting process.  A typical interface to such a counter might be as simple as:
```vhd
entity counter is
port (
    clk_i : in std_ulogic;
    start_value_i : in unsigned;
    start_i : in std_ulogic;
    done_o : out std_ulogic
) 
end entity;
```

Converting every internal counter to an external interface of this form (and it probably needs some expansion for resets etc) is doable but pretty invasive.  Unfortunately there are other sources of hidden state, simple examples include:

* Converting edges to single tick pulses
* Registering outputs

These two can be worked around by running the Device Under Test for an extra tick after every event, but many firmware implementations involve running data and control through a sequence of registers.  Indeed _any_ pipelined operation will be broken by this attempted optimisation.

In brief, I think this is a niche optimisation that may be achievable for certain blocks, but will be very invasive to implement and difficult to get right.

coretl, 2025-04-04:
Discussed this with @Araneidae and we came up with a better idea. At the moment, a block can respond to internal events (based on timestamp counters) as well as external inputs and registers changing:

[image]

The idea we came up with is taking anything that makes internal events (like the timestamp counter) and repackage it as an entity that can be wired as an input to the logic:

[image]

coretl, 2025-04-04:
> entity counter is
> port (
>     clk_i : in std_ulogic;
>     start_value_i : in unsigned;
>     start_i : in std_ulogic;
>     done_o : out std_ulogic
> ) 
> end entity;

These names are better than the ones in the diagram.

> Converting every internal counter to an external interface of this form (and it probably needs some expansion for resets etc) is doable but pretty invasive. 

I think we should try this out for the pulse block and see how far it gets us

> Unfortunately there are other sources of hidden state, simple examples include:
>
>     * Converting edges to single tick pulses
>
>     * Registering outputs
>
>
> These two can be worked around by running [...]

I think we can get round most of them with a "how many ticks do I run for on an event change". Most blocks are 1 or 2, PCAP would be more like 4 due to the pipelining. The counter extraction should deal with PULSE and SEQ, and we only care about soft blocks and PCAP here, so maybe we'll be ok with our limited scope?

> In brief, I think this is a niche optimisation that may be achievable for certain blocks, but will be very invasive to implement and difficult to get right.

I'm hoping our unit tests will catch any subtleties this introduces, but yes we need to test this carefully

coretl, 2025-11-17:
@EmilioPeJu please could you give a brief update on progress and the current strategy in this ticket

EmilioPeJu, 2025-11-17:
I've been working on a testtarget that will be the base for this issue, this target contains all the common soft blocks(as per include file) and pcap. I have already a basic implementation in which I managed to test register accesses to simple blocks like bits and they were working fine. My next step is getting a simple pcap capture simulation working and a simple pgen simulation too.
Those three simulated cases (register accesses, pcap and pgen) will form a good base as system tests that could be run by CI (covering a lot of entities that were not covered in the timing tests and interactions). Then, I will use this test harness to generate a realistic timing sequence of the documented cases in pandablocks.org documentation, this will be the case that I will use to asses the optimization you proposed.