PandABlocks/PandABlocks-FPGA #235: Study PandABox2 Schematics and create constraints file

tomtrafford, 2025-07-09:


glennchid, 2025-07-10:
I already did this, here:
https://gitlab.com/xserra/panda_v2/-/blob/main/Hardware/Mercury_pin_constraints/PandA2_pins.tcl
The net names here match those on the schematics, and they will require some renaming and grouping for use in the PandABlocks framework.

tomtrafford, 2025-07-10:
I don't seem to have access to that link... Do you know if permissions are needed for that repo?

glennchid, 2025-07-10:
Hi Tom, yes you would need permission, but I can add a copy of the file to
this ticket

glennchid, 2025-07-11:
Pin constraints here (zipped as tcl not allowed!):

[PandA2_pins.zip](https://github.com/user-attachments/files/21181316/PandA2_pins.zip)

glennchid, 2025-07-31:
Schematics here (may not be the very latest version):

[board-CoreDesign.pdf](https://github.com/user-attachments/files/21531230/board-CoreDesign.pdf)